INTI-CMNB/kicad_auto #2: ERROR:Time-out waiting for eeschema, giving up

Simsalaba, 2021-02-15:
Running the following command:

`export SUBDIR=kicad-ci-test-spora/pcb_main
export WORKDIR=test-repos
docker run --rm -it -v /tmp/.X11-unix:/tmp/.X11-unix -e DISPLAY=$DISPLAY     -v $(pwd)/$WORKDIR:/home/$USER/workdir     --user $USER_ID:$GROUP_ID     --env NO_AT_BRIDGE=1     --workdir="/home/$USER"     --volume="/etc/group:/etc/group:ro"     --volume="/home/$USER/.config/kicad:/home/$USER/.config/kicad:rw"     --volume="/home/$USER/.cache/kicad:/home/$USER/.cache/kicad:rw"     --volume="/etc/passwd:/etc/passwd:ro"     --volume="/etc/shadow:/etc/shadow:ro"     setsoft/kicad_auto:latest /bin/bash -c "cd workdir/$SUBDIR; kiplot"`

Produces:

ERROR:Output from command:
> ../src/common/stdpbase.cpp(62): assert "traits" failed in Get(): create wxApp before calling this
> ERROR:Time-out waiting for eeschema, giving up (eeschema_do - eeschema_do:156)
>  (kibot.kibot.kiplot - kiplot.py:146)
ERROR:eeschema_do returned 10 (kibot.kibot.out_pdf_sch_print - out_pdf_sch_print.py:56)

I pulled the kicad_auto image and I am running https://github.com/INTI-CMNB/kicad-ci-test-spora

Environment: Debian 10 with XFCE4 desktop on Digital Ocean

set-soft, 2021-02-17:
Hi @Simsalaba !
I suggest:
1) Update the `kicad_auto` image. The one you are using looks outdated (the KiCad message: `assert "traits" failed` is now supressed)
2) Update the Spora example. 
3) Instead of running `kiplot` run `kibot -vv -d Fabrication -s run_drc -i 2> error.log`

This will save a lot of information in `error.log`. Then look in the `Fabrication` subdir. You should get a video with the session named `run_erc_eeschema_screencast.ogv`.

Include both files here.

Simsalaba, 2021-02-19:
[log-and-video.zip](https://github.com/INTI-CMNB/kicad_auto/files/6008811/log-and-video.zip)

It seems it is getting stuck on the KiCad "running for the first time" splash-screen.

set-soft, 2022-06-06:
Hi @swissbyte !
Please describe your setup and how do you get the issue.