INTI-CMNB/kicad_auto #5: 'kicad' Segmentation fault when not setting user to root

mdeweerd, 2021-11-11:
I am working on setting kibot  up under windows. 
The 'kibot" script is provided further below.

I installed VcXsrv .  When in interactive mode, I can run  `kicost`  and I get the `kicost` UI from the docker VM, but when running  `kicad` as a non-root user, I get a `Segmentation fault` .  When setting user and group to `root`, I get the kicad UI.

I tried both dev and the release mentioned in the script.

I run `make KIBOT=./kibot` where kibot is in the current directory.

I tried the `kibot_variants_arduprog` demo project.
I always end up with an interactive shell, which may be my script (I am working on it).

The core issue is the Segmentation fault for a non-root user in the Docker VM.

Script (with root as user, not the current user name):

```bash
#!/bin/bash
IMAGE=setsoft/kicad_auto:10.3-5.1.5
#IMAGE=setsoft/kicad_auto:dev
export USER_ID=$(id -u)
export GROUP_ID=$(id -g)
USER_ID=root
group_ID=root
export WORKDIR=$(realpath "$(dirname $0)/..")
export SUBDIR=$(basename $0)
export WORKDIR=$(realpath "$(dirname $0)")
export SUBDIR=.
export DISPLAY=${DISPLAY:=0}
export HOMEDIR=$(realpath "$(dirname $0)/..")/kibot

# In case of windows batch
# for /f "delims=[] tokens=2" %%a in ('ping -4 -n 1 %ComputerName% ^| findstr [') do set NetworkIP=%%a
# echo Network IP: %NetworkIP
IP=$(ping -4 -n 1 ${COMPUTERNAME}|head -2|tail -1|perl -p -e 's/^.*?\[(.*?)\].*/$1/;')
DISPLAY=$IP:0.0

mkdir -p $HOMEDIR
WORKDIR=$(cygpath -w "$WORKDIR")
HOMEDIR=$(cygpath -w "$HOMEDIR")
#ETC=$(cygpath -w "/etc")
#echo $ETC
ETC=/etc
docker run --rm -it -v /tmp/.X11-unix:/tmp/.X11-unix -e DISPLAY=$DISPLAY \
    -v $WORKDIR:/home/$USER/workdir \
    --user $USER_ID:$GROUP_ID \
    --env NO_AT_BRIDGE=1 \
    --workdir="/home/${USER}" \
    --volume="${ETC}/group:/etc/group:ro" \
    --volume="${HOMEDIR}/.config/kicad:/home/$USER/.config/kicad:rw" \
    --volume="${HOMEDIR}/.cache/kicad:/home/$USER/.cache/kicad:rw" \
    --volume="${ETC}/passwd:/etc/passwd:ro" \
    --volume="${ETC}/shadow:/etc/shadow:ro" \
    $IMAGE /bin/bash
```

set-soft, 2021-11-11:
Hi @mdeweerd !

I don't know how Docker works on Windows. The first thing that comes to my mind is to check the owner and permissions for `/home/$USER/.config/kicad`, `/home/$USER/.cache/kicad`, `/home/$USER/`, etc.

If KiCad can't create files it will most probably die with a `Segmentation fault`, I reported some bugs about the lack of checks in KiCad.

The fact that you can run it as root suggests this is a permission issue. Also: the docker images can be used to run KiCad interactively only for debug purposes, but the normal usage is running all as root. You basically provide a copy of your project (i.e. doing a `git clone`) and just don't worry about using `root`.

I understand that when you run it interactively and share your `$HOME` you won't want to run it as root, but again, the interactive stuff is just to debug things, not for normal KiBot runs.

Is interesting for running old KiCad versions without installing them in the host system.

mdeweerd, 2021-11-11:
Ok, its just that the ['run.sh' ](https://github.com/INTI-CMNB/kicad_auto/blob/master/run.sh)  script provided gets the user id and group and propagates that. 

/home/${USER} is owned by root:root and only writable by root.  `.config` and `.cache` are also owned by root but writeable for everybody (including subdirs).

Anyway, I (more or less) understand the reason:
- Docker uses a virtual machine running on WSL2 and runs its own VMs inside that machine;
- The windows user does not exist there;
- Mapping the passwd and shadow files from cygwin (where the user exists) is not helping because of the above item;

One would need to create the user on the host vm to ensure access rights.

The solution is therefore to run as root and to not map the passwd and shadow files.

Il'll remove that from my script and update in my initial post.

(Now I still have to find how 'kibot' should be run with the docker VM).

mdeweerd, 2021-11-11:
Closing this as the cause is clear, the solution is to run this as root on windows.

Now going to check kibot which is another docker image based on this one... .

closed 2021-11-11 by mdeweerd

mdeweerd, 2021-11-11:
I updated the script again, when setting the user to root, the home directory has to be /root, not /home/<USER> .

I got the demo working - I am going to propose a change there.