INTI-CMNB/kicad_auto #7: Unrecognizeable error

G-Pereira, 2022-03-26:
I am getting this error, any ideas?

The CI code is [here](https://github.com/index-machines/index/blob/af34c6530ba160d678aca913afa2b24b9197cc41/.github/workflows/generate-mobo-pcb-artifacts.yaml#L28)

```
2022-03-25T23:57:34.8950947Z WARNING:(W037) Field Datasheet of component J9 (USB_A Bottom Camera) contains extra spaces: ` ~` removing them. (kibot - v5_sch.py:1029)
2022-03-25T23:57:34.8952180Z WARNING:(W037) Field Datasheet of component J10 (USB_A AUX USB) contains extra spaces: ` ~` removing them. (kibot - v5_sch.py:1029)
2022-03-25T23:57:34.8953192Z WARNING:(W037) Field Datasheet of component J11 (USB_A Top Camera) contains extra spaces: ` ~` removing them. (kibot - v5_sch.py:1029)
2022-03-25T23:57:34.8959092Z WARNING:(W037) Field JLCPCB of component R20 (R_Small 5.1k) contains extra spaces: ` C27834` removing them. (kibot - v5_sch.py:1029)
2022-03-25T23:57:34.8960153Z WARNING:(W037) Field JLCPCB of component R22 (R_Small 5.1k) contains extra spaces: ` C27834` removing them. (kibot - v5_sch.py:1029)
2022-03-25T23:57:34.8961087Z WARNING:(W037) Field JLCPCB of component R28 (R_Small 120) contains extra spaces: ` C17437` removing them. (kibot - v5_sch.py:1029)
2022-03-25T23:57:34.8962600Z WARNING:(W037) Field LCSC of component R28 (R_Small 120) contains extra spaces: `C131390 ` removing them. (kibot - v5_sch.py:1029)
2022-03-25T23:57:34.8964586Z WARNING:(W037) Field Mouser of component U14 (MAX3078E) contains extra spaces: `700-MAX3078EESA ` removing them. (kibot - v5_sch.py:1029)
2022-03-25T23:57:34.8971425Z - Running the ERC
2022-03-25T23:57:46.5985840Z WARNING:(W058) Missing KiCad main config file /github/home/.config/kicad/6.0/kicad_common.json (eeschema_do - misc.py:177) (kibot - kiplot.py:152)
2022-03-25T23:57:46.5986505Z WARNING:(W058) Eeschema failed to start retrying ... (eeschema_do - eeschema_do:628) (kibot - kiplot.py:152)
2022-03-25T23:57:46.5991560Z - Updating BoM in XML format
2022-03-25T23:58:04.0717002Z - 'Interactive Bill of Materials' (ibom) [ibom]
2022-03-25T23:58:04.0757152Z WARNING:(W013) `G***` component in board, but not in schematic (kibot - kiplot.py:288)
2022-03-25T23:58:04.0786186Z WARNING:(W013) `svg2mod` component in board, but not in schematic (kibot - kiplot.py:288)
2022-03-25T23:58:04.0786772Z WARNING:(W013) `kibuzzard-61025CA6` component in board, but not in schematic (kibot - kiplot.py:288)
2022-03-25T23:58:04.0787300Z WARNING:(W013) `kibuzzard-61046E92` component in board, but not in schematic (kibot - kiplot.py:288)
2022-03-25T23:58:04.0787840Z WARNING:(W013) `kibuzzard-61046EDB` component in board, but not in schematic (kibot - kiplot.py:288)
2022-03-25T23:58:06.1998185Z - 'Bill of Materials' (bom) [bom]
2022-03-25T23:58:06.2585247Z - 'HTML BOM' (html_bom) [bom]
2022-03-25T23:58:06.3132118Z - 'Print schematic (PDF)' (print_sch) [pdf_sch_print]
2022-03-25T23:58:16.5660130Z ERROR:Eeschema created an error dialog, expect a fail (eeschema_do - eeschema_do:66) (kibot - kiplot.py:149)
2022-03-25T23:58:16.5661026Z ERROR:Eeschema failed to start or reported a fatal error (eeschema_do - eeschema_do:633) (kibot - kiplot.py:149)
2022-03-25T23:58:16.5661514Z ERROR:eeschema_do returned 10 (kibot - out_pdf_sch_print.py:73)
2022-03-25T23:58:16.7098333Z ##[error]Process completed with exit code 12.
```

set-soft, 2022-03-26:
Hi @G-Pereira !
As the error message says KiCad couldn't load the file and opened an error dialog.
To know what's the error you should run KiBot with a high level of verbosity (-vvv).
I suggest trying to generate only the problematic output and capture all the data generated. The most important thing is an OGV video with the KiCad session, in this video you'll be able to see the error dialog.

set-soft, 2022-03-26:
I found the problem. Was fixed by: INTI-CMNB/KiBot@637a6917c08da2b7c00f2800adc02d6567fe74db

The dialog said:
[image]

closed 2022-03-26 by set-soft

set-soft, 2022-03-26:
BTW: Is quite annoying to checkout the `index` project.

Using 6000x4000 pixel images for docs doesn't look like a good idea.  The documentation dir weights around 640 MiB and looks like the images changed with the time because the repo weights 2 GiB. Lamentably is too late to fix it without losing the history. But in the future consider using a separated repo for the docs and also using images in smaller resolution. 

Just saving the images with 85% quality (instead of 97%) reduces them 3.3 times! And the same pictures saved in 3000x2000 @ 85% weight less than 10% the original size. This could make the docs folder to be less than 70 MiB, and the repo less than 210 MiB. But now is too late. 

Note: I understand you didn't commit those files (was Stephen), but I just wanted to comment it to someone involved in the project.

G-Pereira, 2022-03-28:
I appreciate the feedback!