INTI-CMNB/kicad_auto #9: No zip?

thejpster, 2022-05-05:
The container doesn't seem to include /usr/bin/zip. This makes it difficult to create custom Zip files of the build outputs in a pipeline.

(In my case I wanted to rename some of the outputs, but only when building a release against a pushed tag, and them zip them).

Would you be OK with a PR to add it?

set-soft, 2022-05-06:
Hi @thejpster !
The container is mainly oriented to KiBot use. Using KiBot you have zip, rar and tarball creation, using the `compress` outputs.
Do you really need the zip command?

thejpster, 2022-05-07:
Well, maybe?

What I’m trying to do is make a zip file of my Gerber outputs with a given name. What I’d then like to do (but don’t do currently) is make it so that if you unpack archive X, you get a folder called X, with the files inside.

See https://github.com/Neotron-Compute/Neotron-Pico/blob/c5afe9ce787dae4ec73d9cb1fe7a64b63ea311f1/.github/workflows/kibot.yml#L103

I’ll go and look at the kibot docs to see if it can do that internally.

thejpster, 2022-05-07:
OK, the `compress` output is good enough. Thanks!

closed 2022-05-07 by thejpster