INTI-CMNB/kicad_auto #20: kidiff installation seems to be broken

N0cynym, 2024-05-02:
Currently I use `kicad8_auto:latest` and my GitLab pipeline fails very early because the docker image seems to be broken.

I install into the docker some specific tools and get the following result:

```
$ apt install something_else
WARNING: apt does not have a stable CLI interface. Use with caution in scripts.
Reading package lists...
Building dependency tree...
Reading state information...
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 kidiff : Depends: kiauto but it is not installable
          Recommends: scour but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```

set-soft, 2024-05-02:
Hi @N0cynym !

This is true for *dev* images where *kiauto* is from the GitHub repo.

N0cynym, 2024-05-02:
Thanks for this information will commit 26229f1db2e7cd96b2313f80f5e15f93467a531c solve this for future cases or should `apt --fix-broken install` be keeped in local pipeline files?

set-soft, 2024-05-02:
I verified that the current *ghcr.io/inti-cmnb/kicad7_auto_full:dev* doesn't have this problem, the others should also be fixed.

closed 2024-05-02 by set-soft