INTI-CMNB/kicad_auto #21: [Feature] Include a fixed (not yet released) interactive html bom (ibom)

nguyen-v, 2024-10-15:
**Describe the bug**

When generating the interactive HTML bom output, some of the Kibuzzard components on the silkscreen are misplaced:
https://github.com/EPFLXplore/XRE_LeggedRobot_HW/blob/kibot/amulet_controller/Generated/Manufacturing/Assembly/amulet_controller-ibom.html
When generating manually using the action plugin, I get the correct output:
https://github.com/EPFLXplore/XRE_LeggedRobot_HW/blob/kibot/amulet_controller/Manufacturing/Assembly/ibom/ibom.html

**To Reproduce**

`git clone git@github.com:EPFLXplore/XRE_LeggedRobot_HW.git`

`cd XRE_LeggedRobot_HW/`

`git checkout kibot`

`cd amulet_controller/`

`kibot -c kibot.kibot.yaml`

**Expected behavior**

Generated HTML BOM should match the one generated with the action plugin (no misplaced silkscreen)

**Screenshots**
Misplaced:
[image]
Some are also placed outside of the board for some reason:
[image]
[image]

Correctly placed:
[image]

[image]

**Environment (please complete the following information):**

WSL2 on Windows, running Ubuntu 24.04 LTS. Using the `dev` branch which I installed with 
`pip3 install --user -e .`
According to https://kibot.readthedocs.io/en/latest/installation.html

nguyen-v, 2024-10-15:
Hi @set-soft,
you are correct, it seems like I generated the HTML bom back when the project was still under KiCad 7. I tried uninstalling/reinstalling HTML BOM through PCM, and generating it manually results in the same misplaced silkscreen

set-soft, 2024-10-15:
Don't worry, is fixed upstream, I'll do a maintenance release of ibom for KiBot docker images.

set-soft, 2024-10-15:
Here is the package:

https://github.com/INTI-CMNB/InteractiveHtmlBom/releases/tag/v2.9.0-2

How did you install iBoM in your WSL2+Ubuntu system? Did you use the Debian package?

nguyen-v, 2024-10-15:
Probably not ideal, but I first installed it according to:
https://set-soft.github.io/debian/
Then, because I wanted to be able to use the last dev branch locally, I uninstalled kibot with
`sudo apt-get remove kibot`
And installed the dev branch with:
`pip3 install --user -e .`

Should I just do:
`sudo apt install interactivehtmlbom.inti-cmnb_2.9.0-2_all.deb`
to get the latest version?

set-soft, 2024-10-15:
> Then, because I wanted to be able to use the last dev branch locally, I uninstalled kibot with `sudo apt-get remove kibot` 

I usually have the stable release installed at system level and run the *dev* version straight from the repo. Kibot can run like this, no need to install. You just need to invoke the *src/kibot* script from the repo. You can create a convenient symbolic link to it.

> Should I just do: `sudo apt install interactivehtmlbom.inti-cmnb_2.9.0-2_all.deb` to get the latest version?

Yes

nguyen-v, 2024-10-15:
Got it!
Everything is working fine now, thanks!