INTI-CMNB/kicad_auto #23: apt broken in kicad9_auto image

xanthio, 2025-05-04:
It's not possible to install extra package on kicad9_auto image as tools have broken dependency with KiCAD.

Kicad is correctly installed in version **9.0.1**
```
# kicad-cli --version
9.0.1
```

```
# apt upgrade              
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 interactivehtmlbom.inti-cmnb : Depends: kicad (>= 5.1.0) but it is not installed or
                                         kicad-nightly but it is not installable
 kiauto : Depends: kicad (>= 5.1.0) but it is not installed or
                   kicad-nightly (>= 5.1.0) but it is not installable
          Recommends: ssvnc but it is not installed
 kibot : Depends: kicad (>= 5.1.6) but it is not installed or
                  kicad-nightly but it is not installable
 kicost : Depends: kicad but it is not installed or
                   kicad-nightly but it is not installable
 kidiff : Depends: kicad (>= 5.1.0) but it is not installed or
                   kicad-nightly (>= 5.1.0) but it is not installable
          Recommends: scour but it is not installed
 kikit : Depends: python3-shapely but it is not installed
         Depends: kicad (>= 6.0.0) but it is not installed or
                  kicad-nightly (>= 6.0.0) but it is not installable
         Depends: kicad-footprints but it is not installed or
                  kicad-nightly-footprints but it is not installable
         Depends: kicad-symbols but it is not installed or
                  kicad-nightly-symbols but it is not installable
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```

The proposed command would result in the installation of the broken packages - We want them.

I noticed a change in the ki9 Dockerfile 
`dpkg --ignore-depends kicad -i kicad-git*.deb`

I guess it is linked to the fact that on base image kicad/kicad:9.0 kicad is not installed as a package. Why ?

set-soft, 2025-05-05:
> I guess it is linked to the fact that on base image kicad/kicad:9.0 kicad is not installed as a package. Why ?

Yes this is a consequence of using KiCad official images. Mark just does a "make & install", is easier, but creates this problem.

> I noticed a change in the ki9 Dockerfile

Yes, this is the workaround.

Now that Debian is catching-up I'll see if I can create images using Debian packages

xanthio, 2025-05-05:
I managed to build an image based on debian:bookworm-slim where I installed Kicad from backport to get the latest kicad version and then installed kibot deb package from release.

Not fully tested yet but hope this would help you improving your image.